spacecraft-design-lab-2019/groundtruth-simulator #26: usage of `Sensor` object

tomerarnon, 2019-11-14:
@hridujain @kevincrispie I know you've both been working on the sim, so pinging you both here.

https://github.com/spacecraft-design-lab-2019/groundtruth-simulator/blob/a97c972359a15eb8fdca3ca6de8a02d02b3ee646/Simulator/sensors.py#L40-L51

This was not the intention of the `Sensor` class. Each`Sensor` should only be constructed once. With this function (and the rest of them in this file) each time it is called, it allocates a new sensor object and then samples from it once. 

The point of the `Sensor` class is that it is an object that keeps track of the covariance, noise model, bias, etc. of the sensor. Hard coding those deep into a hot function rather than at top level is pretty silly (and will be wasteful).

Instead, the sensors should live in the spacecraft. E.g.:
`spacecraft.sensors = {"sun": Sensor(...), ...}` or 
`spacecraft.sun_sensor = Sensor(...)` if you prefer, etc.. 
Whichever paradigm makes the most sense (I'd probably go with the explicit case for our purposes, since there's only ~3 sensors). In the sim, you will call `spacecraft.mag(...)` or `spacecraft.sun_sensor(...)` or just `spacecraft.sense(...)` or something like that. Again, whatever makes the most sense.

hridujain, 2019-11-14:
Hi Tomer, thanks for the note. Totally see what you mean about reconstructing the sensor class repeatedly. This was definitely an oversight. Will restructure this code soon.

tomerarnon, 2020-01-09:
Looks like this was resolved, so closing.

closed 2020-01-09 by tomerarnon