spacecraft-design-lab-2019/groundtruth-simulator #32: Environment class at propagator level

tomerarnon, 2019-11-15:
https://github.com/spacecraft-design-lab-2019/groundtruth-simulator/blob/8fd486fabfc19372682a7407691c30bed23e3bd7/Simulator/propagate_step.py#L79

@hridujain github blames you for the line above 😜 

This was also not the intention of the `Environment` class (see #26 for more details) the environment should also be a "top level" object. Basically anything that is "persistent" in real life should be persistent in the sim.

hridujain, 2019-11-15:
Haha darn it! Github caught me. We talked about this in the meeting today. Changes are incoming. We will no longer be reinitializing any classes

https://github.com/spacecraft-design-lab-2019/groundtruth-simulator/blob/8fd486fabfc19372682a7407691c30bed23e3bd7/Simulator/propagate_step.py#L79

tomerarnon, 2020-01-09:
This appears to have been resolved. `Environment` is still imported here:
https://github.com/spacecraft-design-lab-2019/groundtruth-simulator/blob/3058550dca1aaaec70e3621c535fb250b6e58a9c/Simulator/propagate_step.py#L6

but that can be removed without consequence, and isn't really a big problem anyway.

closed 2020-01-09 by tomerarnon