spacecraft-design-lab-2019/groundtruth-simulator #33: random walk gyro bias

tomerarnon, 2019-11-15:
This issue is in reference to the meeting notes from earlier that say "Add gyro bias to state vector". 

I _strongly_ disagree with that approach if it is at all avoidable. Basically I think it would be ideal if we could keep the state as just the literal state `(r, q, v, w)` to the best of our abilities. I've come up with one possible way of dealing with this, which is to implement a `RandomWalk` class and plant it in it's own errormodel (as an alternative or subclass of the `LinearErrorModel`) and use it as the intrinsic bias vector. I've implemented something I think could work below. The bottom line idea is that it would hook into `sensor.errormodel.b` in the gyro sensor. To handle turning `update` on and off, we would either need to have a boolean flag in propagate_step (and wherever else is appropriate) that tells us if this is a "real update" or an "imagined update", or a spacecraft variable that does the same job that gets flipped back and forth as necessary. Not sure which is cleaner (but leaning towards the spacecraft struct one)

Let me know what you think, if you have other ideas, think it's too complicated, etc.:

```python
class RandomWalk(object):
    """
    Implements the function `state` with keyword `update` which determines whether 
    to increment the state of the random walk. The argument defaults to `False`. 
    If `True` is provided, the state is updated _after_ the state is returned.

    Constructor arguments:

    - `a` = 1; The multiplier on the noise (i.e. the noise amplitude).
    - `state` = 0; The starting state of the random walk.
    """
    def __init__(self, a = 1, state = 0):
        self.a = a
        self._state = state

    def state(self, update = False): 
        out = self._state
        if update:
            self._state += self.a * np.random.rand()
        return out

# Example:
rw = RandomWalk()
print(rw.state())

# In propagate_step, we do rk4 which involves multiple updates that take place 
# "from the same timestep". So in this case we would pass in the (default) argument 
# `update = False` and the  RandomWalk won't update:

print(rw.state())

# Later, we do the real time update, so we would call the updating one (note the update 
# takes place after the returned value):

print(rw.state(update = True))
print(rw.state())
```

hridujain, 2020-01-15:
@tomerarnon we should revisit this issue. You mentioned somewhere else that modifying the LinearErrorModel to add a random walk is incorrect and I agree. Do you think the above is still a good way to implement the random walk on the sim side? 

I also feel strongly against adding it to the literal spacecraft state vector. I believe the current bias can simply live within the `sensors.gyroscope` object.

tomerarnon, 2020-01-15:
The argument in favor of storing it in the state, as I recall, was that it would be needed in the MEKF or something like that and therefore might be easier to carry around everywhere. Is/was that true @pdetrempe? If so, there may be a case for putting it in the state, but even then I'm not so sure.

I still think this could be a decent idea with just a bit of sprucing up.

hridujain, 2020-01-15:
That would be on the on-board side. On the sim side, we can store the bias wherever we'd like, and this will never be passed to the on-board code. The only allowed interfacing between the on-board (including MEKF) and sim is passing spoofed sensor measurements and receiving command inputs.