spacecraft-design-lab-2019/groundtruth-simulator #58: "CaSense" and other names

tomerarnon, 2020-01-09:
@hridujain @kevincrispie I see the names `caSense` and `scaleF` in several places in the code, and think they should change to something more descriptive. I had no idea what they meant until one of you told me. These aren't "standard variables" and there are no context clues with which one could reasonably figure it out. We should rename them to something more along the lines of `crossaxis_sensitivity` and `scalefactor` (or whatever else you prefer). It would be nice if the code spoke for itself. 

Here is one example of it being used (where the variables actually are described!), but of course I mean it should change everywhere. I'm sure it's not the only case of a badly named variable, just one that stood out to me. If you know of others lets get upgrade them too.

https://github.com/spacecraft-design-lab-2019/groundtruth-simulator/blob/3058550dca1aaaec70e3621c535fb250b6e58a9c/Simulator/sensormodels.py#L123-L135

hridujain, 2020-01-09:
I have no problems with making this change. @kevincrispie if you agree than I can make the quick fix.

hridujain, 2020-01-15:
addressed this in branch "sim_tidy".

closed 2020-01-15 by hridujain